alxhoff/FreeRTOS-Emulator #1: How to make it work on macOS after compile?

prsolucoes, 2019-10-25:
Hi,

I have compiled it on macos, but how to make it run?

```
file bin/FreeRTOS_Simulator
FreeRTOS_Simulator: Mach-O 64-bit executable x86_64
```

When i execute it:
```
./FreeRTOS_Simulator 
Failed to match './FreeRTOS_Simulator' with '(.+bin)', returning 1
FreeRTOS_Simulator(29935,0x10835a5c0) malloc: can't allocate region
*** mach_vm_map(size=4160427126381547520) failed (error code=3)
FreeRTOS_Simulator(29935,0x10835a5c0) malloc: *** set a breakpoint in malloc_error_break to debug
getBinFolderPath malloc failed
```

Thanks.

prsolucoes, 2019-10-25:
I make it work on macOS. But its freeze on main screen. I tested on Ubuntu and the same problem, main window is freeze. There is any infinite loop or something like this? It is not working.

Thanks.

alxhoff, 2019-10-29:
Hi @prsolucoes,
i will spin up an Ubuntu machine and give it a test. I'll get back to you.

alxhoff, 2019-11-04:
Hi @prsolucoes 
my quickly written example code had a deadlock in it and the simulator itself is fine. Please let me know if the problem persists.

closed 2019-11-04 by alxhoff