alxhoff/FreeRTOS-Emulator #3: aIOSocketPut assertion

alxhoff, 2020-03-26:
> Connecting to (null):2222 failed
> FreeRTOS_Emulator: /home/alxhoff/git/GitHub/FreeRTOS-Emulator/src/main.c:365: vDemoSendTask: Assertion `!aIOSocketPut(TCP, NULL, TCP_TEST_PORT, test_str_3, strlen(test_str_3))' failed.

Is this what you are also getting @PhilippvK?

PhilippvK, 2020-03-26:
Yeah I think so

alxhoff, 2020-03-26:
Yer i used netcat for testing originally. I'll fix this error sometime, unless you've already solved it?

PhilippvK, 2020-03-26:
Nope, always commented it out.

Might be trivial, but have you already checked if 0 is really returned on success?

PhilippvK, 2020-03-26:
Take your time to fix this, I think even for the Enemy-Implementation we‘ve been talking about we could live with a RX-only connection. (I am just trying out to write a UPD sender (in C) for the left paddle of your pong game.)

alxhoff, 2020-03-26:
hahahah cool :)

> Nope, always commented it out.
> 
> Might be trivial, but have you already checked if 0 is really returned on success?

Haven't looked at it yet. I'll put a decent amount of time aside next week to polish the code base.

alxhoff, 2020-03-27:
@PhilippvK Have found that the error comes from an interrupted sys call, still in the process of solving it. May have drunk too much wine to solve it tonight, but found this read and thought it was pretty great. 

http://www.madore.org/~david/computers/connect-intr.html

alxhoff, 2020-03-27:
Wow.....i totally handled this when implementing the IO.....or at least partially. Ops.
[UDP](https://github.com/alxhoff/FreeRTOS-Emulator/blob/0b9c2d07be9ba132cc5d7f699275080dcd96d732/lib/AsyncIO/AsyncIO.c#L432) vs [TCP](https://github.com/alxhoff/FreeRTOS-Emulator/blob/0b9c2d07be9ba132cc5d7f699275080dcd96d732/lib/AsyncIO/AsyncIO.c#L497)

PhilippvK, 2020-03-27:
I just patched my AsyncIO.c and am still getting the same assertion...
Have you also tried it yourself? Maybe I am doing anything wrong...

alxhoff, 2020-03-28:
Shit, you're right. Take two. This time with coffee instead of wine.

alxhoff, 2020-03-28:
Hahahaha my "debug" code was what was fixing it. I was checking errno after fails and I removed my handling of `EINTR` and then the committed code let this fall through to my default error handling that returned -1 and causes the assertion to fail. Seems like the window frreezes sometimes still but I think this is something else. As now the emulator runs long enough for this error to appear. I believe this error is solved.

PhilippvK, 2020-03-28:
Can confirm that it does work now. Thank you for your effort.

closed 2020-03-28 by alxhoff