alxhoff/FreeRTOS-Emulator #5: Hanging

alxhoff, 2020-03-28:
Hangs randomly, maybe unrelated but moving the mouse out and back into graphics window seems to cause the error.

PhilippvK, 2020-03-28:
No error messages or crashes?

PhilippvK, 2020-03-28:
I would recommend to wait until I have my changes in TUM_Draw and TUM_Events online. Maybe it will change the issue.

alxhoff, 2020-03-28:
No errors sadly, all TCP comms etc keep going, it doesn't crash but just enters a deadlock for the graphic stuff. So I imagine it's some poorly implemented demo code from me :P

alxhoff, 2020-03-28:
Also please give me baby size PRs :P

PhilippvK, 2020-03-28:
I will do my best. Sorry for the delays.

alxhoff, 2020-04-01:
Tried this on another arch machine and got similar issues. Both the other machine and mine are running i3 window manager. What is your machine running in terms of distro and graphical interface/window manager?

alxhoff, 2020-04-01:
Just to double check. @PhilippvK  on your system if you mouse in and out of the window ~10+ times it doesn't hang?

PhilippvK, 2020-04-02:
Cannot reproduce on native ubuntu. Running very very smooth, as the computer is quiet fast and has a decent gpu.

Reproducing inside vms on my mac is tricky as the performance is bad anyway so i cannot tell if such behavior is related to the problem you mentioned.

Gonna try it out inside VMs on Linux/Windows Hosts, soon.

alxhoff, 2020-04-02:
Ok cool. I'm going to send it to my colleagues for them to test and see if anyone else gets it as they all have varying distros etc

PhilippvK, 2020-04-02:
Latest 64 bit Manjaro inside virtualbox also works for me.

alxhoff, 2020-04-02:
strange. I'll send it out to my colleagues and see if any of them have problems. What desktop environments run on your systems? I am using i3 window manager and that could be why

PhilippvK, 2020-04-02:
On ubuntu: Gnome desktop

Inside Manjaro VM: XFCE

Definitely could be related to i3.

alxhoff, 2020-04-02:
i'd say so as on my housemates machine it also didn't work and he runs i3 ontop of pure arch

PhilippvK, 2020-04-02:
Try it again after mergin the opengl context stuff. However odds are low...

alxhoff, 2020-04-02:
fingers crossed

closed 2020-04-02 by alxhoff

alxhoff, 2020-04-02:
Also confirmed fixed on secondary Arch/i3 setup