alxhoff/FreeRTOS-Emulator #6: Parsing of Bin Folder Path fails when using gdbserver

PhilippvK, 2020-03-28:
One of my solutions to connect VSCode with the VM makes use of a gdb server. The execution using it is failing right in the main function: 

```
Failed to match 'FreeRTOS_Emulator' with '(.*)/', returning 1
n_

```

It seems like you only accept `argv[0]` to habe at least one forward slash `/` like common with `./FreeRTOS_Emulator` and `../bin/FreeRTOS_Emulator`, but in my case it is simply `FreeRTOS_Emulator`. Thus the REGEX is not satisfied.

The fix in `getBinFolderPath` should be trivial, but I would appreciate if you would take care of it.

alxhoff, 2020-03-28:
I shall do it tonight as I sip on my beer

PhilippvK, 2020-03-28:
Many thanks. No hurries.

PhilippvK, 2020-03-29:
Works flawlessly.

closed 2020-03-29 by alxhoff

PhilippvK, 2020-03-31:
Unfortunately I found another bug with the new implementation.

If you have
`#define LOGO_FILENAME "../resources/freertos.jpg"`

and run the executable inside the bin directory, the bin_path would be `.` making the concatenated path of the Image `.../resources/freertos.jpg`. 

I think there is a `/` missing.

alxhoff, 2020-03-31:
ok. I will fix this up. There are a few gross problems like this floating about as I went for quick and dirty at the time.

PR #8 (2020-03-29): Replaced use of regex for finding bin dir with POSIX dirname, should …
…fix #6

@PhilippvK try this
commit 78b5a49b (2020-03-29): Replaced use of regex for finding bin dir with POSIX dirname, should fix #6
commit c1d763e5 (2020-03-31): Replaced use of regex for finding bin dir with POSIX dirname, should fix #6