alxhoff/FreeRTOS-Emulator #7: VS code integration documentation in README

alxhoff, 2020-03-28:
@PhilippvK Just thought i'd leave a reminder here that we should document the usage of VS code in the README.

PhilippvK, 2020-03-28:
I was going to provide a whole set of manual pages in this Repos Wiki.

PhilippvK, 2020-03-28:
I havent looked into the tracing stuff you implemented yet. Could you possibly write a short manual on how to use it in the following weeks.

Also do you see any way integrating the tracing results into the IDE?

alxhoff, 2020-03-28:
Ahhh right, forgot about that. I implemented that ages ago to help me debug some issue. Hopefully my bash history still has teh right commands :P, will document it

alxhoff, 2020-03-28:
> I was going to provide a whole set of manual pages in this Repos Wiki.

Sound good. I'd just add a link from the readme or just add a note that there are wiki pages. So students don't miss it

PhilippvK, 2020-03-28:
When I do the wiki, I probably also move the YCM (vim) related part of the readme there. And yes i will link it.

alxhoff, 2020-03-28:
Sounds like a good idea

PhilippvK, 2020-04-01:
I also have a tool to make a HTML/PDF export of the Wiki.

alxhoff, 2020-04-01:
a pdf export would be useful for students I think

closed 2020-04-15 by alxhoff