alxhoff/FreeRTOS-Emulator #18: Restructured Error Handling misses SDL_GetError() output

PhilippvK, 2020-04-01:
I really like your error handling using `goto` statements and labels to make code less redundant but by printing errors just via `PRINT_ERROR()` instead of `logSDLError()`, you do loose the information on what did specifically fail in case of working with SDL_CreateWindow() and so on.

This information is provided by `SDL_GetError()` which only called from `logSDLError()`.

Do you have an idea on how to incomporate it?

Targets cdfa82be5f183b0c826e6833b00a443e8a751c06

alxhoff, 2020-04-01:
I created two ,,,,urm..."sub-defines"?

```
#define PRINT_TTF_ERROR(msg, ...) PRINT_ERROR("[TTF Error] %s\n" #msg, (char *) TTF_GetError(),  ##__VA_ARGS__)
#define PRINT_SDL_ERROR(msg, ...) PRINT_ERROR("[SDL Error] %s\n" #msg, (char *) SDL_GetError(),  ##__VA_ARGS__)
```

That now give results such as

> [ERROR] [TTF Error] Couldn't open ./../resources/fonts/IBMPlexSans-Medium.ttf2
> "Opening font @ '`�$��U' failed"    @-> /home/alxhoff/git/GitHub/FreeRTOS-Emulator/lib/Gfx/TUM_Draw.c:628, vInitDrawing
> [ERROR] Failed to intialize drawing    @-> /home/alxhoff/git/GitHub/FreeRTOS-Emulator/src/main.c:579, main

The d-lux in error messages

closed 2020-04-01 by alxhoff

PhilippvK, 2020-04-01:
I really like it. Premium.