alxhoff/FreeRTOS-Emulator #21: Implement a proper way of Task-save Debug messages

PhilippvK, 2020-04-01:
Printing via printf in parallel from multiple task can lead to problems such aus crashes.

I think it would be more optimal to have a single task which prints out messages send to a queue by other tasks inspired by the implementation in:

https://github.com/coon42/FreeRTOS-Posix-Simulator/blob/master/FreeRTOS_Posix/main.c

PhilippvK, 2020-04-01:
I would only do this for verbose (debug) statements. Error handling should be fine with printf, in my opinion.

alxhoff, 2020-04-01:
Agreed. Thread safety is a bit of a mess at the moment I think.

PhilippvK, 2020-04-01:
I won't tackle it before the weekend. Toolchain is more important at the moment. If you want to implement it before, let me know.

alxhoff, 2020-04-01:
I think i'll need to start recording the lectures :/

PhilippvK, 2020-04-01:
Oh no.. Are you using OBS for the screencast?

alxhoff, 2020-04-02:
No idea what i'm going to use yet. I'll check OBS out though.

PhilippvK, 2020-04-02:
Might be a bit overkill as it is used by professional streamers etc. but should also work on linux.

alxhoff, 2020-04-16:
I think i will implement this over the weekend, introducing....TUM_error

PhilippvK, 2020-04-16:
Are you going to have a main periodic print task collecting strings from other tasks via a queue if available?

alxhoff, 2020-04-16:
yep, sounds like a good way to structure it. Although that would make it FreeRTOS dependent. I might just use an unwrapped pthread but yer same underlying idea I think.

closed 2020-07-03 by alxhoff