alxhoff/FreeRTOS-Emulator #25: Internal frame limiting

alxhoff, 2020-04-02:
In the TUM_Draw backend we should add a frame limiter because it just kills the CPU. I doubt the performance in a VM is good enough that we can use the classic "limit FPS" in the project, so I don't think it would hurt to hide a frame limit outside of main.

PhilippvK, 2020-04-02:
So you want to limit the framerate to 60 fps in the TUM_Draw Api and leave the frame rate limiter inside main() for lowering it to f. e. 50 frames (20 ticks)

Would be similar to the setup on real hardware where the screen itself caps at 60 Hz.

alxhoff, 2020-04-02:
> Would be similar to the setup on real hardware where the screen itself caps at 60 Hz.

Good point!

alxhoff, 2020-04-02:
Can't tell if it's limiting to 60, looks like 30 for me because the FPS flickers so much. Might try and smooth the FPS a bit more.

PhilippvK, 2020-04-02:
Flickering was also present on the real hardware when setting the frame limit to 60 fps. In my project I used a moving average to smooth out the framerate but i wanted to keep it as simple as possible in the demo.

alxhoff, 2020-04-02:
yer, i added a moving average but the fps is ~30 with my code. I've made a "i've drunk too much wine" mistake somewhere....

PhilippvK, 2020-04-02:
In a VM or native performance? Dedicated GPU?

alxhoff, 2020-04-02:
found the error and now it's around 55 which is probably a rounding error, time to throw in some floats

closed 2020-04-03 by alxhoff