alxhoff/FreeRTOS-Emulator #33: Add coverity to travis CI

alxhoff, 2020-04-03:


PhilippvK, 2020-09-16:
I think for the future it would be a good idea to move from Travis CI to Github Actions.

alxhoff, 2020-09-17:
Agreed, would be keen to check out GH Actions as well. RIP Travis CI

PhilippvK, 2020-09-17:
I have recently ported another project from Travis to Github. I can give it a try here as well sometime later this month.

alxhoff, 2020-09-17:
be my guest!

PhilippvK, 2020-09-20:
Feel free to have a look at https://github.com/alxhoff/FreeRTOS-Emulator/runs/1140789690?check_suite_focus=true and https://github.com/alxhoff/FreeRTOS-Emulator/blob/ci/.github/workflows/main.yml.

Getting the Sanitizers running with actual Display and Audio Device was the trickiest part...

A few things are still subject to change but here is a list of remaining TODOs:
- [ ] Add error handling to Sanitizers (ASAN is printing out some warnings while USAN, TSAN seem to run almost silently)
- [ ] add coverity target as mentioned in this issue. Maybe this will help: https://github.com/ruby/actions-coverity-scan/blob/master/.github/workflows/coverity-scan.yml
- [ ] disable Travis CI for this repo before merging the `ci` branch into `master`. Else, there may occur conflicts regarding the GitHub Pages deployment.

alxhoff, 2020-09-21:
wow. great stuff! I like the look of the GH actions CI scripting, looks a bit simpler than Travis. I also agree about Travis CI, if I should disable it just let me know. Also the coverity thing is something i've been meaning to do for a while. Will need to write some actual tests or maybe a test fork with code that runs all functions of the TUM libs etc.

PhilippvK, 2020-09-21:
I agree. But the thing which bothers me the most with GitHub Actions is that you can not have shared job-depencies f.e. to install the SDL2 packages.

This would on make the YAML file even simpler and also reduce the runtime if they are only installed once.

alxhoff, 2020-09-21:
That's annoying