alxhoff/FreeRTOS-Emulator #34: Crashes when run from root directory

alxhoff, 2020-04-03:
> [ERROR] [SDL Error] Couldn't open /home/alxhoff/git/GitHub/resources
> "Failed to load image"    @-> /home/alxhoff/git/GitHub/FreeRTOS-Emulator/lib/Gfx/TUM_Draw.c:339, loadImage

alxhoff, 2020-04-03:
Hardcoded image paths, need to decide if this should be how it should be or if we provide some other method.

PhilippvK, 2020-04-03:
This sucks, way to much struggles for such a “simple” feature.

alxhoff, 2020-04-03:
yer i don't think i'd be bothered but just mention somewhere that relative paths mean that you need to be careful where you run it from. Or could pass the resources folder in as a cmake where it could be resolved into an absolute path and passed to make as a define.....maybe i'll be motivated to do that at some point.

PR #44 (2020-04-17): Added error handling for drawing of images that have invalid filepath…
…s, fixes #34

pew pew handle all da errorz:gun:
commit 9baac6a2 (2020-04-17): Added error handling for drawing of images that have invalid filepaths, fixes #34
PR #76 (2020-07-03): Added ability to load resources from the resources folder via file se…
…arch as well as being able to give relative and absolute paths

Should actually fix #34 when done
closed 2020-04-17 by alxhoff