alxhoff/FreeRTOS-Emulator #50: [BUG] Pause State broken in Pong Demo

PhilippvK, 2020-04-19:
**Describe the bug**
2 Bugs:
1. Pause is canceled immediately as keys are not debounced
2. Top and Bottom Walls are flickering in Paused mode as only one buffer contains walls.

**To Reproduce**
Press P

**Expected behavior**
Steady Switches between Game and Paused mode

**Screenshots**
/

**Desktop (please complete the following information):**
Irrelevant

**Additional context**
In #48  if "fixed" the first (debounding) problem by adding a `vTaskDelay(100)` statement, but the problem will still be there if the key is held too long

alxhoff, 2020-05-08:
I will look into this soon as well as updating the branch to use the new refactored code

alxhoff, 2020-05-08:
I also think that 1 is completely ok and doesn't really need addressing. Just those walls.

PhilippvK, 2020-05-08:
> I also think that 1 is completely ok and doesn't really need addressing. Just those walls.

But it gets worse if you comment out my vTaskDelay workaround...

PhilippvK, 2020-05-08:
> I will look into this soon as well as updating the branch to use the new refactored code

Are you going to move all drawing statements to the PongControlTask?

I think this would solve the flickering as the double framebuffer is not a problem when it is cleared every frame.

alxhoff, 2020-05-08:
yer I can do that

alxhoff, 2020-05-12:
Took the easy path and just blanked the screen during pause (except text). Seems like pressing pause a few times can freeze up the game. Will have to look into that first before this is closed.

alxhoff, 2020-05-12:
All fixed up, also added randomness to ball starting angles (velocities)

closed 2020-05-12 by alxhoff