alxhoff/FreeRTOS-Emulator #53: [IMPROVEMENT] Remove old branches to make things easier

PhilippvK, 2020-04-19:
**What should be improved?**
We have a lot of temporary development branches which are not required anymore.

**Describe the solution you'd like**
The minimal set of branches would be  `master` and `pong`. I can live with some more, but 9 are too much.

**Additional context**
Add any other context or screenshots about the improvement here.

PhilippvK, 2020-04-19:
I already deleted some of my old branches. The ones left are mostly related to pen pull requests. Lets just make sure that we do not forget to remove them when the merge is complete

alxhoff, 2020-04-20:
The ones that are there now are from my stuff, mainly the persistent TCP #20 and the safe print #21. gh-pages is used by travis ci for uploaded generated doxygen documentation to the repo's pages page.

closed 2020-04-20 by alxhoff