alxhoff/FreeRTOS-Emulator #56: [FEATURE]

PhilippvK, 2020-04-29:
**Is your feature request related to a problem? Please describe.**
You can not change the Font easily or use multiple fonts in one program.

**Describe the solution you'd like**
_Best Option:_ Implement a way to change fonts on the fly.

**Describe alternatives you've considered**
At least allow overwriting the default font (by using `#ifndef DEFAULT_FONT ...` in `TUM_Draw.h`.

alxhoff, 2020-04-29:
good idea

PhilippvK, 2020-04-29:
Similar situation exists for the Waveforms. At the moment they only can be defined in the libraries itself or am I wrong?

alxhoff, 2020-04-29:
yer i think so. i'll look into a solution. Maybe some sort of auxiliary init function that takes in a custom font/audio

PhilippvK, 2020-04-29:
Hm, I would prefer to be able to use more than one font (and font size) during program execution. Do not feel forced to implement it, if its to complex. Its just a nice to have thing.

alxhoff, 2020-04-29:
i'll make a font API now

closed 2020-04-30 by alxhoff