alxhoff/FreeRTOS-Emulator #61: [IMPROVEMENT] Cache images between frames

PhilippvK, 2020-05-06:
**What should be improved?**
Drawing larger images or many small ones reduces the framerate heavily, even when running on native Linux. Caching the files could be worth checking out.

**Describe the solution you'd like**
Do not destroy the texture/surface between every frame as this should hopefully reduce the overall load.

**Additional context**
Maybe make this optional. For uGFX there is a method which enables the caching on a per-image basis.

This could also reduce the performance impact of scaling images if the scaled version of the texture is caches instead of the original, but them you would need to check if the scale factor has changed...

alxhoff, 2020-05-06:
Do you remember if calling `SDL_CreateRenderer` when binding GL context was necessary? Or is the important call `SDL_CreateRenderer`?

PhilippvK, 2020-05-06:
Do you mean `SDL_GL_MakeCurrent`?

I am very sure, that the renderer has to be created after claiming the context which needs to be done in the thread performing the drawing operations...

Why are you asking?

alxhoff, 2020-05-06:
`SDL_GL_MakeCurrent` is what i meant. Problem is that loaded images textures are tied to a renderer and if the renderer is being potentially recreated then the image textures become invalid (i think).

PhilippvK, 2020-05-06:
You can try out moving it, but you probably would also try to move the context claiming which likely breaks the drawing stuff again..

alxhoff, 2020-05-06:
the context claiming only needs the window and doesn't actually reference the renderer which is why I was wondering if they are actually connected.

PhilippvK, 2020-05-06:
I know it is not a good habit to do such things, but if we assume, that a context will only be created one, it should be fine?

alxhoff, 2020-05-06:
yer I don't really want to make this assumption but I can't find documentation that links the context and the renderer

PhilippvK, 2020-05-06:
Alternatively, you could let the user init all the used images in main, so the caching could be done in vBindDrawing each time after the renderer was created.

alxhoff, 2020-05-06:
yes but this would require keeping a list of all loaded images etc which would be cumbersome

PhilippvK, 2020-05-06:
I think creating the renderer without a context resulted an SDL error or null pointer.

PhilippvK, 2020-05-06:
Try it out..

PhilippvK, 2020-05-06:
I also think the complexity of this is high, given the fact that we do not know if it improves the performance or if it is going to introduce other issues.

PhilippvK, 2020-05-07:
I am very impressed. Improvements are definitely recognizable (f.e. when Drawing the Logo >10 times).

In my VM I sometimes experience crashed when the Emulator is launching. If have the feeling that that it gets worse the more graphics are initialized in the `main()` but I would ignore it for now. Happened for the 4th time in 2 week now.

closed 2020-05-07 by alxhoff