alxhoff/FreeRTOS-Emulator #66: [BUG] Hanging event fetch

alxhoff, 2020-05-29:
**Describe the bug**
@duclos-cavalcanti exercise submissions hang after about ~2minutes and the backtrace has thread 23 hanging on an SDL event fetch

@duclos-cavalcanti could you please commit your exercises and link the commit here?

[backtrace.txt](https://github.com/alxhoff/FreeRTOS-Emulator/files/4700626/backtrace.txt)

alxhoff, 2020-05-29:
Possibly [this](https://github.com/duclos-cavalcanti/FreeRTOS-Emulator/commit/c8895c37a93dd941e0240c846af7b1929809ad96) commit @duclos-cavalcanti?

alxhoff, 2020-05-29:
Removing excess calls to `tumDrawBindThread` in non-frame-swapping tasks seems to of fixed the issue. @PhilippvK we should add concurrency protection to `tumDrawBindThread`.

alxhoff, 2020-05-29:
Further testing shows that it still happens but takes longer.

alxhoff, 2020-05-29:
Appears there are multiple threads calling `tumDrawUpdateScreen`, aside from the GL context issue here I think this should also be locked.

duclos-cavalcanti, 2020-05-29:
Hi,

is this happening with the zip uploaded to moodle? The branch was specified
in the .md file I believe, and it would be the last commit in that branch
the latest version. Would you want me to make a new commit fixing that bug
and link it? Im not really sure what my next step should be. On my Manjaro
things seem to work fine. I can test it by letting it hanging for  2 to 3
minutes and see what happens.

BR,

Daniel

PhilippvK, 2020-05-29:
If it is uses somewhere outside the FrameSwapper, the FPS limit does not work anymore.

alxhoff, 2020-05-29:
@duclos-cavalcanti try leaving it to run for a couple of minutes with the calls to `tumDrawBindThread` and `tumDrawUpdateScreen` removed everywhere outside of the swapper task and see if it still hangs. Just hung for me even after removing those calls.

@PhilippvK I don't follow?

alxhoff, 2020-05-29:
> Hi, is this happening with the zip uploaded to moodle? The branch was specified in the .md file I believe, and it would be the last commit in that branch the latest version. Would you want me to [...]

Also don't worry as far as your submission goes. All good there. Just wanting to see if this is an emulator problem that should be solved asap.

duclos-cavalcanti, 2020-05-29:
> @duclos-cavalcanti try leaving it to run for a couple of minutes with the calls to `tumDrawBindThread` and `tumDrawUpdateScreen` removed everywhere outside of the swapper task and see if it still hangs. Just hung for me even after removing those calls.
> 
> 
> 
> @PhilippvK I don't follow?
> 
> 

Doing it right now, ill reply with the result.

alxhoff, 2020-05-29:
Also seems to happen on the second screen.

duclos-cavalcanti, 2020-05-29:
Removed tumdrawbindthread to the two other tasks that were using it and also removed tumdrawupdatescreen to the only other task that was calling it apart from vswapbuffers.

Im letting it just run, and should I notify if it crashes after 2-3 minutes? I can also film the screen if that helps. (with a proper desktop app)

duclos-cavalcanti, 2020-05-29:
> Possibly [this](https://github.com/duclos-cavalcanti/FreeRTOS-Emulator/commit/c8895c37a93dd941e0240c846af7b1929809ad96) commit @duclos-cavalcanti?
> 
> 

this is the correct/latest commit from Exercises_END.

duclos-cavalcanti, 2020-05-29:
5 minutes passed with the cleaner version. Nothing happened as far as crashing or freezing. Should I try to press all the functionalities during run time to force the program to draw as much as possible?

alxhoff, 2020-05-29:
hmm strange. Probably some sort of edge case OpenGL with my desktop environment (or lack there of). I will have more of a look into it later. Only seems to happen with your submission.

duclos-cavalcanti, 2020-05-29:
Did Philipp run it on his computer? That could answer the question of it being something to do with your setup.

duclos-cavalcanti, 2020-05-29:
I commited the “cleaner” version.

alxhoff, 2020-05-29:
> Did Philipp run it on his computer? That could answer the question of it being something to do with your setup.

Not yet. @PhilippvK

PhilippvK, 2020-05-29:
It crashes for me, too. Reproducible by pressing a button after running the program for a few minutes

alxhoff, 2020-05-29:
Can you paste a backtrace here?

duclos-cavalcanti, 2020-05-29:
any button? or the “a” button? I can try that again.

alxhoff, 2020-05-29:
I believe I have experienced this now with a separate submission. :/

alxhoff, 2020-05-29:
@PhilippvK does yours crash if you remove all the `tumFetchEvents` calls except the one in the swapper?

PhilippvK, 2020-05-29:
Fixes the issue for me.

PhilippvK, 2020-05-29:
I moved the function there inside the frameswapper as it is not thread save an should not run in parallel to the opengl screen update. It is likely that it will also crash if multiple fetchEvents are executed in parallel. But as events are fetched in the frameswapper anyway I do not understand why it is used in every task.

duclos-cavalcanti, 2020-05-29:
Ok... I’ll take the unnecessary functions out.

alxhoff, 2020-05-29:
I think i'll update teh backend to make all of the GL context calls thread safe. @PhilippvK do you know if the SDL event fetching is dependent on the GL context? This will make it a bit crap as we'd have to keep track of GL bound threads and then temporarily take and then restore context if events are to be polled in more than one task.

PhilippvK, 2020-05-29:
Using the GLContext would be preferable but i can not provide any proof that the events need it.

duclos-cavalcanti, 2020-05-29:
I understood the general explanation of why the program would crash when running fetchEvents and UpdateScreen in a parallel manner. But just for the sake of curiosity, I tested it out again, where I would let it run for 2- 5 minutes without pressing anything and then would press whatever buttons necessary to observe the program's reaction. 

When I made a command outside of the program, like Alt+Tab to change windows, I noticed it crashed. 

However, when I waited 5 minutes on a stopwatch then began pressing the "in-game"buttons to see if it would have any effect, like "F", "A", "B" and so on; the program ran normally. Does my home setup influence it in any way?

I'm more so asking, so I have a better understanding when going into the final project. That way I can protect myself from dangerous mistakes.

alxhoff, 2020-06-02:
To be honest I don't think I can explain the exact behaviors as that would require me to go into the backend of the SDL libraries in quite depth. Usually with these race events they can work 99 out of 100 times and then they randomly crash.

closed 2020-06-02 by alxhoff