alxhoff/FreeRTOS-Emulator #74: [FEATURE] Emulator config

alxhoff, 2020-06-29:
**Is your feature request related to a problem? Please describe.**
Changing "variables" such as SCREEN_HEIGHT and possible enabling of features such as FPS limit could be set in a FreeRTOSConfig style file.

**Describe the solution you'd like**
A `EmulatorConfig.h` that allows for user build control

@PhilippvK please help me is listing possible config options that would be useful

So far I have:

- Screen sizes 
- Window title
- Default font
- fonts directory
- resources directory
- enabling fps limit

PhilippvK, 2020-06-29:
Except 
- Specify Target framerate

I do not really have other ideas...

alxhoff, 2020-06-29:
thanks. I am sure people will mention things in the future

closed 2020-06-30 by alxhoff